simondankelmann/Bluetooth-LE-Spam #259: Need help

WinException, 2024-03-21:
I want to edit swift device names or add more but i can't find it on src

simondankelmann, 2024-03-21:
You can find and edit them here:

https://github.com/simondankelmann/Bluetooth-LE-Spam/blob/main/app/src/main/java/de/simon/dankelmann/bluetoothlespam/AdvertisementSetGenerators/SwiftPairAdvertisementSetGenerator.kt

after that, you have to delete the apps data and cache, this ensures that the database is recreated with the items you added :)

WinException, 2024-03-23:
i have question bcs i see option connectable what does it do?

simondankelmann, 2024-03-23:
> i have question bcs i see option connectable what does it do?

that is something you can set in android, propably if you want to advertise a legitimate device that some other device can connect to. i didnt dive deeper into this but changing it could cause the popups to not appear

WinException, 2024-03-23:
btw how to fix apple ads, i want to troll my friend

KornineQ, 2024-04-05:
how to you make categories in swift?

closed 2024-04-29 by simondankelmann