simondankelmann/Bluetooth-LE-Spam #261: A lot of packages being sent, but mostly nothing appears on iphones.

SashaLore, 2024-03-29:
I set advertisement duration to 2000ms, turned on 2 devices list, but nothing appears on my classmates iphones. On iphones from xr to 11 only works apple tv thing and autofill, but on my friend's iphone 13 works some kind of devices (but not all of them). Also I tried 20ms and checked it with 2nd phone with spam detector, and it showed that I was sending 20-40 packages/sec ig, but almost all of them are being ignored by iphones.

KaspersP, 2024-03-30:
I found the solution for some reason it only works in the evening so you have to set a manual time like 20:00

KaspersP, 2024-03-30:
Also use the ios action module list and be within a close range

Alluseri, 2024-04-02:
> I found the solution for some reason it only works in the evening so you have to set a manual time like 20:00

On the receiving device or on the sending device?

PatrikPepega, 2024-04-06:
Same problem

fillwithjoy1, 2024-04-06:
> I set advertisement duration to 2000ms, turned on 2 devices list, but nothing appears on my classmates iphones. On iphones from xr to 11 only works apple tv thing and autofill, but on my friend's iphone 13 works some kind of devices (but not all of them). Also I tried 20ms and checked it with 2nd phone with spam detector, and it showed that I was sending 20-40 packages/sec ig, but almost all of them are being ignored by iphones. (1st device - Honor 9X global 4/128, 2nd device - Honor 7a Pro 2/16 rooted)

You know, it is possible that Apple has probably fixed this in the latest version of iOS....

Commander-Crash, 2024-04-08:
Apple has parched this in the latest ios17 update. It will still display but will not display more then once if it detects spamming of ble. Android has patch a similar issue as well, as far as samsung ble spam works on Samsung still.

PooChild, 2024-04-25:
> Also use the ios action module list and be within a close range

this helped me so much! I changed the time to 20:00 or 8:00PM and it worked like a charm! Also, I kind of figured out the action moduals are the only ones that work myself lol. These are good tips and I can comfirm they work! Don't know if I would've figured out to change the time by myself!!!

closed 2024-04-29 by simondankelmann