simondankelmann/Bluetooth-LE-Spam #282: On Xiaomi Poco M6 doesn't work

Ciucea, 2024-06-13:
I have android 14, Xiaomi HyperOS 1.0.3.0 and when i open the app, it s automatically crashed. Can someone help me?

[image]

StingerFingersinmyass, 2024-06-13:
Grant all necessary permissions to the application. If it doesn't work, clear the data or install the debug version and grant all necessary permissions again. Closed.

Ciucea, 2024-06-13:
> Grant all necessary permissions to the application. If it doesn't work, clear the data or install the debug version and grant all necessary permissions again. Closed.

Now it is working. Thank you!

closed 2024-06-21 by simondankelmann