simondankelmann/Bluetooth-LE-Spam #297: [Feature Proposal] Automatic Update Integration from GitHub Releases

CodeWithTamim, 2024-08-10:
Hi
I hope you're doing well. I’m considering adding a feature to the app that would allow it to show an update dialog when a new release is available on GitHub. The feature would also handle downloading and installing the update directly from within the app.

Would you be open to this enhancement? Please let me know if this is something you'd like to include.

Thank you!

Best regards,
Tamim Hossain

simondankelmann, 2024-08-10:
Hi :) sounds interesting in general, at the moment i don't see any reasons for major new releases to come though. So if you want to spend your time on this, feel free but I am not sure if it will be ever used within this app 😅

Anyway thanks for the proposal!

CodeWithTamim, 2024-08-10:
> Hi :) sounds interesting in general, at the moment i don't see any reasons for major new releases to come though. So if you want to spend your time on this, feel free but I am not sure if it will be ever used within this app 😅
> 
> Anyway thanks for the proposal!

I saw people use this at college for fun regularly. And i researched on the update mechanisms so i can writeit in 2/3 hours. Pr coming soon 😅

simondankelmann, 2024-08-10:
Nice thx, take your time. I won't be able to check this next week anyway

CodeWithTamim, 2024-08-10:
Okay 😅👍

closed 2024-08-10 by CodeWithTamim