simondankelmann/Bluetooth-LE-Spam #308: Add google pixel

Bloxproject, 2024-10-16:
It would be cool to hjack my own google pixel and doing experience of a range of how far to my Galaxy phone using this app and how far can it affect the giigle pixel 9 8 7 6 or any to trick the pixel phone that it has a new accessory. And I do this for my phone.

acidvegas, 2024-10-17:
> It would be cool to hjack my own google pixel and doing experience of a range of how far to my Galaxy phone using this app and how far can it affect the giigle pixel 9 8 7 6 or any to trick the pixel phone that it has a new accessory. And I do this for my phone.

what the fuck did I just read lol.....

unknown1soldier, 2024-10-25:
> what the fuck did I just read lol.....

agree with you 😂

ha2ked, 2024-10-25:
@Bloxproject If you could try rephrasing this, I am having trouble understanding you.

closed 2024-11-04 by ha2ked