simondankelmann/Bluetooth-LE-Spam #352: Remove extra line breaks in app description on F-Droid

Korb, 2025-01-28:
Please remove unnecessary line breaks:

[image]

thgoebel, 2025-01-30:
A lot of apps on F-Droid have this issue. I’d say this is an issue with F-Droid not handling Markdown nicely/properly.

Korb, 2025-01-30:
Yes, this problem exists in many apps, but many apps have solved this problem. The problem is not that there is something insurmountably preventing us from making the app description good, but that the F-Droid platform's handling of files is not obvious.

For example, your full_description.txt looks like this:

```
This project focuses on utilizing the built-in Bluetooth Low Energy (BLE)
functionality of Android smartphones to create Phantom Bluetooth Device Advertisements,
similar to what is known, for instance, in the case of the Flipper Zero. While there
are other apps available that provide similar functionality, the objective of this
app is to enhance convenience and user-friendliness in the process.
```

All you have to do is change it by removing the line breaks so that F-Droid stops inserting a line break tag at the end of each of your lines, and the description will look worthy of your app.

thgoebel, 2025-01-30:
But that results in long lines, which is 1. annoying to use in some text editors and 2. results long git diffs when edited that are much harder to read.

Imho F-Droid should treat this like Markdown. In Markdown, too, single linebreaks are ignored. E.g.  compare the raw README with how Github renders it. I would prefer to fix this upstream in F-Droid rather than have all 3 to 4K apps on F-Droid needing to “fix” this.
https://github.com/simondankelmann/Bluetooth-LE-Spam/blob/main/README.md?plain=1

Korb, 2025-01-30:
F-Droid has been discussing this for eight years or more. I doubt they'll make any progress. F-Droid developers are too... _slo-o-ow_.

Will your app description look just as ugly if you wrap it in `<p>` without changing the existing line breaks?

Have you considered hosting your app on third-party repositories for F-Droid that have different app metadata requirements?

thgoebel, 2025-01-30:
This has nothing to do with the repository, this is purely a client rendering issue.

closed 2025-01-30 by Korb