simondankelmann/Bluetooth-LE-Spam #361: Not sending LE requests

KailJ1, 2025-03-12:
Hello, I encountered a problem that on phones only 1-2 notifications about connecting devices are displayed, after which they simply stop appearing. I constantly encounter this error regardless of the device model (Android and Apple). What should I do?

KoustuvGanguly, 2025-03-12:
Facing same issue.

DebugBoard, 2025-03-12:
If you are using Android Spam then that is expected as it has been patched all other spams should work though.

KailJ1, 2025-03-12:
Well, I checked on many phones, both Android and Apple, and they all have the same problem. There are 2-3 notifications, after which nothing comes.

DebugBoard, 2025-03-12:
Could I get a video please?

KailJ1, 2025-03-14:
Unfortunately, I can only provide a video from an Android platform. The video will show two phones: one where the program is launched and the other where notifications appear on the screen.

DebugBoard, 2025-03-14:
Okay thats fine

DebugBoard, 2025-03-19:
Problem with the exploit itself not the app therefore out of our control sorry.

closed 2025-03-19 by DebugBoard