simondankelmann/Bluetooth-LE-Spam #365: Release version signed with debut cert

ail1020, 2025-03-23:
It seems the release version is signed with the android debug certificate. It should be signed with a developer certificate.

DebugBoard, 2025-03-23:
My knowledge of this is very limited could you send some docs?

ail1020, 2025-03-23:
https://developer.android.com/studio/publish/app-signing

For reference I downloaded the app-release.apk file from the github releases and scanned it with MobSF to see that it was signed with `Issuer: CN=Android Debug, O=Android, C=US`. Though you can probably also see the signer certificate information if you were to upload the apk to virustotal.com as well.

DebugBoard, 2025-03-23:
I see what the issue is now. All apks from actions or our pre releases will have a debug cert as there isn't the proper key to build a release cert in GitHub actions.

ail1020, 2025-03-23:
https://proandroiddev.com/how-to-securely-build-and-sign-your-android-app-with-github-actions-ad5323452ce

DebugBoard, 2025-03-23:
Okay thanks, just going to ask the owner if he's okay with it.

DebugBoard, 2025-03-24:
Yup he said it's okay to go ahead with it so I'll start tonight.