simondankelmann/Bluetooth-LE-Spam #375: Feature request: edit name of windows swift pair devices

Mrunknown-da, 2025-04-17:
These device names like device 1, device 2 and so on are only placeholders and could be edited. The name is stored in the manufacturer data as a hex data. I made the name Trojan32 for example: it's 0x030080**54726F6A616E3332**. 
The name part of that is the **bold** one, which can be changed individually (the words have to be Encoded in hex obviously). The rest can and should remain as it is.

Thank you for the nice project.👍

DebugBoard, 2025-04-22:
See the issue is the first time the App starts those items will be saved in the database and later used while transmission and I haven't figured out how to make it save to db on chainging a device name. So I have tried, made a ui for it but couldn't get it to work though. I could make a work around but it won't be out soon.

DebugBoard, 2026-09-29:
Implemented in #455

closed 2026-09-29 by DebugBoard