simondankelmann/Bluetooth-LE-Spam #381: App crashes when I select "Fast Pair"

OlegXio, 2025-05-06:
App crashes when I select "Fast Pair".
https://github.com/user-attachments/assets/061cdf40-4f0e-4786-a6ce-528d3d122e07

DebugBoard, 2025-05-06:
Try the release APK from [this build](https://github.com/simondankelmann/Bluetooth-LE-Spam/actions/runs/14700771903).

OlegXio, 2025-05-07:
The first few times after launch, the application did not dye, but the device broadcast "-", and after several such attempts, the application also began to dye.

OlegXio, 2025-05-07:
Noticed: the application does not dye when Legacy Adversting is enabled, but the device displays as on a screenshot:

[image]

DebugBoard, 2025-05-07:
> Noticed: the application does not crashes when Legacy Adversting is enabled, but the device displays as on a screenshot:
> 
> [image]

That is most likely due to your own personal android settings if the text isn't visible but is the app working otherwise?

OlegXio, 2025-06-09:
Sorry for long wait.
Everything works perfectly with the other device profiles, and notifications also display the text correctly when using those other device types
[image]