simondankelmann/Bluetooth-LE-Spam #382: App crash on startup

sussyboy123, 2025-05-06:
When I installed the app I turned on all the permission but when the database loads the app crash I tried clearing the storage and caches still same problem

DebugBoard, 2025-05-06:
Hmmm could I get a screen recording or log pls?

sussyboy123, 2025-05-07:
> Hmmm could I get a screen recording or log pls?

How do I get the log?

DebugBoard, 2025-05-07:
Using Android Studio Logcat.

sussyboy123, 2025-05-07:
> Using Android Studio Logcat.

Sorry I can't use my computer right now but here's the video https://streamable.com/enqzeh

DebugBoard, 2025-05-07:
Try the release APK from [this build](https://github.com/simondankelmann/Bluetooth-LE-Spam/actions/runs/14700771903). You need to allow location in the background.

sussyboy123, 2025-05-07:
> Try the release APK from [this build](https://github.com/simondankelmann/Bluetooth-LE-Spam/actions/runs/14700771903). You need to allow location in the background.

Yes this build worked and didn't crash thanks

closed 2025-05-07 by sussyboy123

DebugBoard, 2025-05-07:
np